F4HTB/Universal_HamRadio_Remote_HTML5 #2: function takes at most 5 arguments (7 given)

parlotto, 2020-10-25:
Hello,
On a pc runing ubuntu 20.04, I have this error when trying to configure :

 http server started
SERVER port 8888
AUDIO outputdevice plughw:CARD=U0x41e0x30d3,DEV=0
AUDIO inputdevice plughw:CARD=U0x41e0x30d3,DEV=0
HAMLIB rig_pathname /dev/ttyUSB2
HAMLIB rig_model TS590S
HAMLIB rig_rate 38400
HAMLIB trxautopower True
**(<class 'TypeError'>, TypeError('function takes at most 5 arguments (7 given)'), <traceback object at 0x7f7e7bde5540>)**

Maybe a problem with the version of hamlib ?
Thanks for this promising software.
Phil.

F4HTB, 2020-10-28:
Can you get more informations please?
Maybe the screen output?

Thanks

parlotto, 2020-10-28:
Bonjour, 
Mon indicatif est F5OWL. 
Merci pour ce soft. 

Voici ce que j'obtiens au lancement : 
philippe@jupiter:~/Universal_HamRadio_Remote_HTML5$ ./UHRR 
(<class 'TypeError'>, TypeError('function takes at most 5 arguments (7 given)'), <traceback object at 0x7f19332679c0>) 
http server started 
SERVER port 8888 
AUDIO outputdevice plughw:CARD=U0x41e0x30d3,DEV=0 
AUDIO inputdevice plughw:CARD=U0x41e0x30d3,DEV=0 
HAMLIB rig_pathname /dev/ttyUSB2 
HAMLIB rig_model TS590S 
HAMLIB rig_rate 38400 
HAMLIB trxautopower True 

j'accède par https://127.0.0.1:8888 
Ensuite quand je fais : "Save and Restart server" 

philippe@jupiter:~/Universal_HamRadio_Remote_HTML5$ (<class 'TypeError'>, TypeError('function takes at most 5 arguments (7 given)'), <traceback object at 0x7f05f234db80>) 
http server started 
SERVER port 8888 
AUDIO outputdevice plughw:CARD=U0x41e0x30d3,DEV=0 
AUDIO inputdevice plughw:CARD=U0x41e0x30d3,DEV=0 
HAMLIB rig_pathname /dev/ttyUSB2 
HAMLIB rig_model TS590S 
HAMLIB rig_rate 38400 
HAMLIB trxautopower True 
(<class 'TypeError'>, TypeError('function takes at most 5 arguments (7 given)'), <traceback object at 0x7fc11c7fb840>) 
http server started 

J'ai installé le soft sur ubuntu 20.04. 
J'utilise hamlib 3.3 issue des paquets de la distribution ubuntu 20.04. 
Le navigateur est firefox 82.0. 
En pj mon fichier de conf. 
En espèrant que ce soit utile. 
73 
Philippe

F4HTB, 2020-11-15:
use raspberry pi instead.
I cannot transpose the code to the stable sources in ubuntu and raspiOS on the same git repo.

closed 2020-11-15 by F4HTB

madpsy, 2021-08-31:
Caused by old version of pyalsaaudio. You probably have v0.8.x installed, you need 0.9.0.

`pip3 install --user pyalsaaudio==0.9.0`