F4HTB/Universal_HamRadio_Remote_HTML5 #6: Unable to turn ICOM IC-7300 rig back on

dslotter, 2020-10-28:
I have one of the most popular ham radio rigs at the current time: an ICOM IC-7300. Many other hams also own this radio.

UHRR is able to turn off this radio, but is unable to turn it back on with the calls to FOO.setPower(1)

By comparison, I can use the rigctl program from HamLib to turn this radio both off and on:

This turns the radio off successfully:
`pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5/opus $ rigctl -m 3073 -r /dev/ttyUSB1 -s 19200 set_powerstat 0`

This turns the radio on successfully:
`pi@raspberrypi:~//Universal_HamRadio_Remote_HTML5/opus $ rigctl -m 3073 -r /dev/ttyUSB1 -s 19200 set_powerstat 1`

dslotter, 2020-10-28:
Here is the debug output for when this issue occurs. Notice the lines in bold -- UHRR attempts to turn the rig on repeatedly but fails.

**Attempting to turn rig on...**
new connection on ControlTRX socket.
getFreq:
Send to all: getFreq:7132000
getMode:
Send to all: getMode:LSB
PING
connection closed for audioRX
connection closed for TX socket
error TXMETER
Shutdown TRX
Shutdown TRX
Shutdown TRX
Shutdown TRX
Shutdown TRX
Shutdown TRX
Shutdown TRX
**Attempting to turn rig off...**
WARNING:tornado.general:SSL Error on 10 ('192.168.1.233', 16725): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
WARNING:tornado.general:SSL Error on 10 ('192.168.1.233', 16726): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
WARNING:tornado.general:SSL Error on 11 ('192.168.1.233', 16727): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
new connection on AudioRXHandler socket.
WARNING:tornado.general:SSL Error on 15 ('192.168.1.233', 16731): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
WARNING:tornado.general:SSL Error on 16 ('192.168.1.233', 16732): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
WARNING:tornado.general:SSL Error on 11 ('192.168.1.233', 16733): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
new connection on AudioTXHandler socket.
**Attempting to turn rig on...**
overrun
new connection on ControlTRX socket.
getFreq:
Send to all: getFreq:0
overrun
getMode:
overrun
Could not obtain the current Mode via Hamlib!
Send to all: getMode:LSB
PING
PING
PING
PING
PING
PING
WARNING:tornado.general:SSL Error on 14 ('192.168.1.233', 16760): [SSL: SSLV3_ALERT_CERTIFICATE_UNKNOWN] sslv3 alert certificate unknown (_ssl.c:1056)
overrun
**Attempting to turn rig on...**
new connection on ControlTRX socket.
overrun
getFreq:
Shutdown TRX
overrun
Send to all: getFreq:0
getMode:

dslotter, 2020-10-28:
This code change works for me, but is obviously clumsy:

        def setPower(self,status=1):
                try:
                        if status:
                                print("Attempting to turn rig on...")
                                self.rig.set_powerstat(Hamlib.RIG_POWER_ON)
# NEW CODE BELOW:
                                print("Attempting to turn rig on using rigctl...")
                                os.system("rigctl -m 3073 -r /dev/ttyUSB1 -s 19200 set_powerstat 1")
# NEW CODE ABOVE:
                        else:
                                print("Attempting to turn rig off...")
                                self.rig.set_powerstat(Hamlib.RIG_POWER_OFF)
                        self.infos["powerstat"] = status
                except:
                        print("Could not set power status via Hamlib!")
                return self.infos["powerstat"]

F4HTB, 2020-10-28:
Hello dear OM, 

Can you say me if you use the packaged hamlib (installed by apt-get install) or if you use the buildead version of last hamlib from the git?

Tha packaged stable version is the 3.2 (if i am not wrong) but the compiled is the 4.1 version. 

You can see on the bottom of the readme that you can get last by:

apt-get remove --purge Hamlib
git clone https://github.com/Hamlib/Hamlib
cd Hamlib
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install

and finaly you have to luch it with  PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
assiming your are in the directory.

Can your try this?

closed 2020-11-02 by F4HTB

dslotter, 2020-11-02:
@F4HTB Hi. I didn't see your GitHub response for some reason until you closed it.

I tried your instructions in the comment above, and it seems to fix this issue. Thanks for your follow-up.

F4HTB, 2020-11-03:
@dslotter 

Hello,
No problem.
I'm glad it was able to help you. I had the same for my 817.
The hamlib has had a lot of improvement between 3 and 4.
Thanks to them!

73!