F4HTB/Universal_HamRadio_Remote_HTML5 #7: List of Radios ? DMR-DStar-Fusion

B-C-C, 2020-10-30:
Not sure if in right spot - feel free to move / redirect

Noticed radio control appears to be done only if radio supports mic keypad

How controlling radio limited in docs but believe covered by above question.

Which radio is being used specifically for testing in project?

Is there a good list of hardware needed exactly..
rpi v3 and up / zero an option?
Add to existing pistar setups possible? Via another port?
USB audio splitter
Supported mobile / base radio models - known?
Ex - MD-9600 , ID-5100A
Cables / Adaptors misc
How many need to be made vs can be bought?

(I probably missed / didn't read well enough)
Didn't notice in readme...

Beyond that I'm really interested in using with DMR unit but will need to add mode option and look for specific tags on how to get that mode added ..

Would figure other digital options would be the same, but don't know for sure..

ntoulasd, 2020-10-30:
Radio control is with Hamlib. Read the wiki for supported radio models.
https://github.com/Hamlib/Hamlib/wiki
Hamlib use CAT interface of Radio not mic keypad.
External (analogue) sound interface is used so there is no problem about any modulation FM, SSB or digital voice.

B-C-C, 2020-10-30:
(Resolved - Feel Stupid)
Thanks!

closed 2020-11-02 by F4HTB