F4HTB/Universal_HamRadio_Remote_HTML5 #10: Reverse proxy config - working great

madpsy, 2020-11-05:
Not sure if anyone else would find this useful but I have UHRR running behind Apache (with authentication). Here's what you need. Replace the proxy IP for your own Pi's address and the IP address in the auth section with whatever address space you want to access without authentication.

Password file can be created with htpasswd -c /var/www/html/.htpasswd-uhrr username

```
SSLProxyEngine On
SSLProxyVerify none
SSLProxyCheckPeerCN off
SSLProxyCheckPeerName off
SSLProxyCheckPeerExpire off
ProxyPass "/CTRX" "wss://192.168.9.35:8888/CTRX"
ProxyPass "/audioRX" "wss://192.168.9.35:8888/audioRX"
ProxyPass "/audioTX" "wss://192.168.9.35:8888/audioTX"
ProxyPass "/" "https://192.168.9.35:8888/"
ProxyPassReverse "/CTRX" "wss://192.168.9.35:8888/CTRX"
ProxyPassReverse "/audioRX" "wss://192.168.9.35:8888/audioRX"
ProxyPassReverse "/audioTX" "wss://192.168.9.35:8888/audioTX"
ProxyPassReverse "/" "https://192.168.9.35:8888/"

<Location "/">
    <If "-R '10.0.0.0/8'">
            Require all granted
    </If>
    <Else>
            AuthType Basic
            AuthName "My Radio"
            AuthUserFile /var/www/html/.htpasswd-uhrr
            Require valid-user
            Require all denied
    </Else>
</Location>
```

F4HTB, 2020-11-06:
Thanks for this.
With your contributions, I wonder if I shouldn't set up a wiki 👍

F4HTB, 2020-11-06:
I registered the addition of an authentication in the to do list.
I am thinking of starting from this and maybe add an external lib allowing the posix, ldap, mysql or other.
It should be easy to find.

https://gist.github.com/guillaumevincent/4771570

closed 2020-11-07 by F4HTB