F4HTB/Universal_HamRadio_Remote_HTML5 #9: RTS is set high on startup

madpsy, 2020-11-05:
I'm using a Yaesu 897 and as soon as UHRR starts it seems to set the RTS (and DTR) serial pins to high, meaning my radio instntly goes into TX mode. I assume this is hamlib doing this under the hood but unsure how, using UHRR config, to stop this behaviour.

F4HTB, 2020-11-05:
Hi, 

The simple way for you it's to désactivate TX on dtr rts in the 897.
I've had these kinds of jokes before.
Rather, it comes from the default states of the driver and in particular in the FTDI chips.
If you want to go further, maybe you can try to compile hamlib 4 as shown on the REAMDE of github.
For my part I will see if it is possible to force the state of the RTS and DTR to the low level in the next update.

Thanks for report.
Get me some news.

73 F4HTB.

madpsy, 2020-11-05:
Thanks for reply.

I switched to a different Serial->USB module and it behaves. I couldn't see an option in the 897's menu to disable the RTS behaviour.

That said, I am keen to try the latest Hamlib anyway, but I'm seeing the following.

It seems /usr/local/lib/libhamlib.so.4 exists but for whatever reason it isn't being found so not sure what's going on there.

`Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory`

F4HTB, 2020-11-09:
@madpsy 

Hello, I come to get some news about your problem.
I have send à message to hamlib team for get news function to set rts or dtr level.

For the last hamlib version have you build and lunch like this:

git clone https://github.com/Hamlib/Hamlib
cd Hamlib
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install

and lunch UHRR like:

PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR

?

madpsy, 2020-11-10:
I still see this

```
pi@rpi4:~/Universal_HamRadio_Remote_HTML5 $ PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory
```

madpsy, 2020-11-10:
This works, though not sure if it loads the latest libs:

```
pi@rpi4:~/Universal_HamRadio_Remote_HTML5 $ PYTHONPATH=/usr/local/lib/python3.7:$PYTHONPATH ./UHRR
recording...
HTTP server started.
new connection on AudioRXHandler socket.
new connection on AudioTXHandler socket.
new connection on ControlTRX socket.
getFreq:
Send to all: getFreq:144300000
getMode:
Send to all: getMode:USB
PING
PING
```

madpsy, 2020-11-10:
It worked after copying libhamlib.so.4:

`cp /usr/local/lib/libhamlib.so.4 /usr/lib/arm-linux-gnueabihf/`

N7JYS, 2020-11-13:
Is there a possibility for getting GPIO support for PTT for routing tx audio thru a data port (Yeasu FT-891). As it sets now tx audio has to be routed thru the mic jack because  PTT is using CAT control. Using rigctl command in Hamlib I can assign a gpio pin for ptt (rigctl -P GPIO -p 25.)  This could feature could be selected in UHRR.conf.  

Thanks

Eric N7JYS

F4HTB, 2020-11-13:
Hi,

Okey, please can you test if it's possible to set it in a pull request 
on the git hub?

Thanks.

73 F4HTB.

Le 13/11/2020 à 15:22, N7JYS a écrit :
> Is there a possibility for getting GPIO support for PTT for routing tx
> audio thru a data port (Yeasu FT-891). As it sets now tx audio has to
> be routed thru the mic jack because PTT is using CAT [...]

N7JYS, 2020-11-13:
Sorry not a programmer but do some hacking on scripts to make changes. I have no experience with PHTYON
However I'm trying to figure out to use your PTT script in UHRR with hamlib rigctl command.   
Eric (N7JYS) Marshall's Communications

closed 2020-11-15 by F4HTB