F4HTB/Universal_HamRadio_Remote_HTML5 #12: Encode TX Clarification

madpsy, 2020-11-06:
I see the alt tag for 'Encode TX' suggests it will enable the Opus Codec (only for TX?).

It should probably be clarified in the README what this is intended for and its advantages/disadvantages and (I'm curious too) why it isn't enabled by default.

F4HTB, 2020-11-06:
Yes indeed, during my first tests, I had indeed taken a code which already uses the opus codec.
However, it did not seem wise to me to part with it because this could improve the latency or the bandwidth during the transmission.
I did not implement it in reception because, as you know, who says encoding means loss of information. Also this seemed to me against indicated for the reception.
As you have surely noticed, for the moment the reception is in 8khz which is more than enough but I have not yet really worked to reduce the emission for technical choices, hence the possible advantage of the codec.
I would also like to see the possibility of sending digital modes etc ... so I keep as is under the elbow.
I am in the process of seeing to implement a system based on RRNs to remove QRMs ... To wait and see.

73!

madpsy, 2020-11-06:
Keeping it uncompressed for data modes is certainly interesting. I have tested access over 4G and with 8 KHz uncompressed this was no problem in terms of bandwidth etc. Latency jumped between 100-400ms which sounds high but it worked perfectly and was impressed at just how well it performed.

closed 2020-11-07 by F4HTB