F4HTB/Universal_HamRadio_Remote_HTML5 #13: Certificate location

madpsy, 2020-11-06:
A couple of points.

1) Providing a pregenerated certificate/key pair as part of the code base isn't ideal. Unless users generate their own certificate and private key there is zero security gain as the key is public knowledge.

2) /boot is most definately not the best place to store them. If anything, keeping them in the same directory as the application is better or even a subdirectory.

3) I question why use encryption at all. The Tornado server itself, as it stands, cannot be Internet facing (not least because of no authentication) and so what do you gain from encrypting the traffic? Having it proxied behind Nginx or Apache is a much better idea for Internet facing use cases, in which case the TLS termination and authentication is handled by them.

I'm happy to fix 2) with a PR and I think 1) should be explained further in the README, which again I'm happy to do if you would like. As for 3) is it a more fundimental change which needs to be considered.

F4HTB, 2020-11-06:
Hello,

I take note of your comments.
The only reason why I had to implement ssl encryption so https is that html5 APIs focusing on resource usage such as the microphone are subject to this restriction.
If it wasn't, I wouldn't have set it up.
The security related to ssl encryption is not a priority here since the messages sent over the air are not themselves encrypted.
The use of a certificate distributed in the git does not therefore have to act as a security vulnerability but as a necessary element linked to the constraints of the APIs except in the countries requiring it.
For example, in France, this can be a complement with respect to the law indicating not to give the opening of flows on a public network and acts as a wall.
Also, the indication of /boot allows those who wish to create a redistributable image of the complete solution to leave the possibility to end users to replace these certificates with little knowledge.

If you want to add more information on this subject maybe in the readme, I thank you in advance for your help. Indeed, my professional situation related to COVID as a computer scientist does not leave me much time to work on it for some time yet. I hope to be able to add new features soon :)

madpsy, 2020-11-06:
Ah ok! I didn't realise there was an API requirement when requesting access to the microphone but that makes perfect sense now.

I'll have a think if I can add anything here.

closed 2020-11-07 by F4HTB