F4HTB/Universal_HamRadio_Remote_HTML5 #15: Console is spammed "overrun" and ui is unresponsive

jvann94, 2020-11-07:
When starting uhhr it starts fine and then the console is spammed constantly with overrun.

On the ui side none of the radio control buttons seem to do anything to the radio and any entered frequency resets back to zero

jslatten, 2020-11-07:
I'm also seeing this behavior.

F4HTB, 2020-11-07:
Hello, 

Hmm ... I don't know how to reproduce this issue ...
What have you put in the configurations?
Can you copy paste for me from the console?

Thanks in advance.

jslatten, 2020-11-07:
Do you just want

> Hello,
> 
> Hmm ... I don't know how to reproduce this issue ...
> What have you put in the configurations?
> Can you copy paste for me from the console?
> 
> Thanks in advance.

Are you talking about the UHRR.conf file? If so, here it is:

pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ cat UHRR.conf 
[SERVER]
port = 8888
certfile = /boot/UHRH.crt
keyfile = /boot/UHRH.key
debug = True

[CTRL]
interval_smeter_update = 0.5
debug = True

[AUDIO]
outputdevice = plughw:CARD=CODEC,DEV=0
inputdevice = plughw:CARD=CODEC,DEV=0

[HAMLIB]
rig_pathname = /dev/ttyUSB0
retry = 5
rig_model = IC7300
trxautopower = True
rig_rate = 38400

F4HTB, 2020-11-07:
Quick question, have you tried another sound card model? I use a creative soundblaster like this one:

https://www.pieces-electro.com/Prix/SOUND-BLASTER-PLAY--CREATIVE-101116.html

jvann94, 2020-11-07:
I've tried both the onboard soundcard and and my usb one, both issues remian

My config is

`[SERVER]
port = 8888
certfile = /boot/UHRH.crt
keyfile = /boot/UHRH.key
debug = True

[CTRL]
interval_smeter_update = 0.5
debug = True

[AUDIO]
outputdevice = plughw:CARD=PCH,DEV=0
inputdevice = plughw:CARD=Microphone,DEV=0

[HAMLIB]
rig_pathname = /dev/ttyUSB0
retry = 5
rig_model = FT857
trxautopower = True
rig_rate = 4800`

jslatten, 2020-11-08:
For reference I don't really want to use a different audio card as the ICOM IC-7300 is connected via USB cable and presents a virtual sound device that provides input/output for the transceiver instead of using your built in sound card and physical audio cables. It's super convenient and allows you to operate your transceiver with literally 1 cable.

jvann94, 2020-11-08:
For my issues I have tried a second soundcard

F4HTB, 2020-11-09:
@jvann94 hi, what's the result of your new test?
I'm looking for someone who has an ic 7300 to see what happens. :)

closed 2020-11-15 by F4HTB