F4HTB/Universal_HamRadio_Remote_HTML5 #16: Audio Lag Problem

dslotter, 2020-11-07:
Kevin Loughin reports on his YouTube channel https://youtu.be/JC9N0PMu_Kc?t=603 that UHRR web interface works pretty well on a laptop, but _with a tablet or smartphone_, when you click the transmit button, talk, and release transmit, there'll be a small pause, and perhaps a small buzz, and _then the audio will lag behind the radio by that amount of time_. This was much more pronounced with the smartphone -- approximately 4-7 seconds behind.

I confirmed this behavior on my Android smartphone -- a Pixel 3a XL running Android version 11

Because of this audio delay, real-time communication is not possible on the smartphone.

Also, running on a Raspberry Pi 3 rather than a 4 also seems to exacerbate this issue.

F4HTB, 2020-11-07:
Hello Dear Dave, 

For be sure to correctly understand, this bug only appears on a Pizero or you noticed this on a pi4?
This indication will help me reproduce the problem.

Thanks in advance

dslotter, 2020-11-07:
I've seen this on a Pi 4. And Kevin Loughin saw this on a Pi Zero. I don't
think the choice of Piatters much except perhaps degree.

The main variable here is using the web interface on a tablet or smartphone.

Does this help explain?

F4HTB, 2020-11-07:
I will try to reproduce this and see.
I'll give you info as soon as possible.

dslotter, 2020-11-07:
Please take your time, kind sir. And did you watch the YouTube video link?
It is a short video...

F4HTB, 2020-11-07:
Yes! thanks to him.
I confess to you that I couldn't think that this little project could have so much interest. :)

dslotter, 2020-11-07:
That's what I thought about HamPi. I thought maybe 10 people would download a Ham Radio software image for a Raspberry Pi... 25,000 downloads later...

My GitHub project: https://github.com/dslotter/HamPi

I plan to add this project to HamPi as well... That's how important your software has become!

F4HTB, 2020-11-15:
Hi, I have try on multiples devices. Problem sometime appear and sometime not.
I continue to search and have a big modification to past audio handler on wome javascript workers.
So i close in the interval. :)

closed 2020-11-15 by F4HTB