F4HTB/Universal_HamRadio_Remote_HTML5 #17: OVERRUN Error

N7JYS, 2020-11-07:
After starting UHRR and watching in run in terminal, I issue a command from the web interface to select a new frequency and SSB mode, this is what I observe,

setFreq:0050000000
Send to all: getFreq 5000000
setMode:AM
overrun

Also its taking 12 seconds after sending this command for the web interface to react. And audio is dropping out.

Noticed on a request to connect to webserver  tornado.general gives an SSL error (SSL: sslv3_ale)

Eric

N7JYS, 2020-11-09:
I have notice the same issue even after the updates to UHRR and console.js files. The changes that were made did not fix the overrun issue and the 12 seconds of no audio while changing modes. Also LSB mode sets Upper Side Band on my FT-891.

Other wise that program is a blast and almost there. Thanks

Eric

F4HTB, 2020-11-15:
this error appears when the sound card does not give enough chuncks.

closed 2020-11-15 by F4HTB