F4HTB/Universal_HamRadio_Remote_HTML5 #19: ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

Xpertti, 2020-11-21:
pi@UHRR:~/Universal_HamRadio_Remote_HTML5 $ PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory
pi@UHRR:~/Universal_HamRadio_Remote_HTML5 $

Rasbperry 4 Pi B 4GB

What went wrong?

F4HTB, 2020-11-21:
Hello, 

The message is clear:
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

you use the path for the compiled library of hamlib, did you compile this one or you use the one from the official package?

Xpertti, 2020-11-22:
Hi,

Also with the official package . Same error in installation both ways.

cd /tmp
git clone https://github.com/Hamlib/Hamlib
cd Hamlib
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install

cd /temp
wget https://sourceforge.net/projects/hamlib/files/hamlib/4.0~rc2/hamlib-4.0~rc2.tar.gz/downloads
ls
tar -zwvf hamlib-4.0~rc2
clear
ls
cd hamlib4.0~rc2
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install

I will try today once again.

Best regards,

OH4RL/Pertti

Xpertti, 2020-11-22:
I got this,
pi@UHRR:~ $ cd ~/
pi@UHRR:~ $ git clone https://github.com/F4HTB/Universal_HamRadio_Remote_HTML5.git
Cloning into 'Universal_HamRadio_Remote_HTML5'...
remote: Enumerating objects: 59, done.
remote: Counting objects: 100% (59/59), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 419 (delta 24), reused 31 (delta 10), pack-reused 360
Receiving objects: 100% (419/419), 3.95 MiB | 2.12 MiB/s, done.
Resolving deltas: 100% (230/230), done.
pi@UHRR:~ $ cd Universal_HamRadio_Remote_HTML5
pi@UHRR:~/Universal_HamRadio_Remote_HTML5 $ ./UHRR
Traceback (most recent call last):
  File "./UHRR", line 19, in <module>
    from rtlsdr import RtlSdr
ModuleNotFoundError: No module named 'rtlsdr'
pi@UHRR:~/Universal_HamRadio_Remote_HTML5 $ PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
Traceback (most recent call last):
  File "./UHRR", line 19, in <module>
    from rtlsdr import RtlSdr
ModuleNotFoundError: No module named 'rtlsdr'
pi@UHRR:~/Universal_HamRadio_Remote_HTML5 $ cd /tmp
pi@UHRR:/tmp $ git clone https://github.com/Hamlib/Hamlib
Cloning into 'Hamlib'...
remote: Enumerating objects: 115, done.
remote: Counting objects: 100% (115/115), done.
remote: Compressing objects: 100% (83/83), done.
remote: Total 39157 (delta 42), reused 80 (delta 32), pack-reused 39042
Receiving objects: 100% (39157/39157), 14.34 MiB | 2.83 MiB/s, done.
Resolving deltas: 100% (31282/31282), done.
pi@UHRR:/tmp $ cd Hamlib
pi@UHRR:/tmp/Hamlib $ ./bootstrap

You must have autoreconf installed to compile hamlib.
Download the appropriate package for your distribution,

You must have automake installed to compile hamlib.
Download the appropriate package for your distribution,

You must have libtool installed to compile hamlib.
Download the appropriate package for your distribution.
pi@UHRR:/tmp/Hamlib $ ^C

closed 2020-11-22 by Xpertti

F4HTB, 2020-11-23:
I have send a new issue in hamlib git about the libhamlib.so.4 problem

73

Xpertti, 2020-11-23:
Thank you for your reply. Now I don´t get the error as   libhamlib.so.4.
cd /tmp
git clone https://github.com/Hamlib/Hamlib
cd Hamlib
sudo apt-get install autoconf automake libtool swig (I had to add this line)
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install
sudo ldconfig
make

rigctl -l
rigctl -m 1036 -r /dev/ttyUSB0 -s 38400 f (Now FT-891 rig reponds frequency)

cd Universal_HamRadio_Remote_HTML5
./UHRR

PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR

Unfortunately I cannot connect with the web-server. (
https://uhrr.local:8888/) or https://192.168.100.110:8888/

I am on the local LAN.

73/Pertti