F4HTB/Universal_HamRadio_Remote_HTML5 #20: Installation issue

f4cib, 2020-11-26:
Hi Olivier & all,
Sorry to open this issue which is certainly related to my low knowledge on linux platform, but:
`./configure --with-python-binding PYTHON=$(which python3)
` reports the following error:
```
checking for swig... no
checking for swig3.0... no
configure: error: SWIG is required to build bindings...

```
I git cloned swig & then had to install some additional dependances to make it (i installed it on /home/swig/ with ./configure --prefix=/home/swig)

so i don't know why it is not found while configuring Hamlib, do i need to install swig on on particular folder ? 
I followed http://swig.org/svn.html for installation.

Moreover, Some other dependances were needed after a fresh rasbian install which were not detailed, but this lead us to try to understand a little what we do and learn new things... Ham definition !;)

So i read, i tried some solutions but now i need someone help to have swig correctly installed and continue installation.

Once again not a problem on the code but on the procedure to have it installed

73 de Franck, F4CIB

F4HTB, 2020-11-27:
Bonjour,

A la vu de votre indicatif, j'imagine que je peux répondre en francais.

Effectivement, je n'ai pas eu le temps de retrouver les packages nécessaires afin de compléter le wiki.
Pas besoin d'installer ceux-ci a partir des git mais les packages officiels semblent suffire.

pouvez vous essayer avec:

sudo apt-get install autoconf automake libtool swig

et me dire si cela est suffisant?

f4cib, 2020-11-27:
Merci Olivier, le Français me va aussi ! 
Je m'étais astreint à écrire en Français pour les utilisateurs non francophones qui semblent assez nombreux.
Cela semble suffire pour le ./configure
Google ne m'avait pas indiqué (pas plus que swig !) que je pouvais faire l'install via un paquet.
Merci, Je poursuis l'installation.
73 de Franck, F4CIB

f4cib, 2020-11-27:
Désolé de continuer à t'embeter, mais la deriere commande censée lancer l'appli passe en erreur:
`PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
-bash: ./UHRR: Aucun fichier ou dossier de ce type`

Le serveur (Apache ?) s'installe via la procédure d'installation ou faut il l'installer ?
Le seul warning que j'ai pu voir était un relinking de Hamlib:
`libtool: warning: relinking '_Hamlib.la'`

Ai-je raté quelque chose, j'ai suivi pas à pas l'install du wiki.

F4HTB, 2020-11-27:
Bonjour, 

Pas d'inquiétude, cela ne me dérange pas.

Concernant le message Aucun fichier ou dossier de ce type, il semble que vous ne soyez pas dans le dossier du logiciel.

normalement lorsque vous vous connecter en ssh ou par une fenêtre terminal vous êtes directement dans votre "home directory" qui est par défaut /home/pi
Vous pouvez vérifier avec la commande "pwd" sans les " qui va vous renvoyer le chemin dans lequel vous êtes.
Sinon pour revenir dans votre répertoire home vous pouvez utiliser "cd ~".
Ensuite il faut télécharger le logiciel.
La procédure complète serait:
```
cd ~
git clone https://github.com/F4HTB/Universal_HamRadio_Remote_HTML5
cd Universal_HamRadio_Remote_HTML5 
PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
```

En espérant que cela puisse vous aider.
Il semble que depuis les dernières mises à jours de raspiOS vous pouvez avoir un message type hamlib.so.4 not found.
Cela vient du fait que l'installation manuelle de la hamlib c'est mal passé a cause de la présence de la version paquet installée par apt-get.
Si c'est le cas vous pouvez faire un sudo apt-get remove hamlib puis dans le dossier hamlib que vous avez télécharger et où vous avez compiler, vous refaire sudo make install puis cd binding puis sudo make install.

Tenez moi informé, je peux certainement vous dépanner à distance si nécessaire.

Cordialement,

Xpertti, 2020-11-27:
You may try this...

cd /tmp
git clone https://github.com/Hamlib/Hamlib
cd Hamlib
sudo apt-get install autoconf automake libtool swig (I had to add this line)
./bootstrap
./configure --with-python-binding PYTHON=$(which python3)
make
sudo make install
cd bindings
make
sudo make install
sudo ldconfig
make

cd Universal_HamRadio_Remote_HTML5
./UHRR

PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR

73

rkiendl, 2020-11-28:
from a plain installation (buster lite) i suffer on finding the module rtlsdr. installing osmocom didn't do the trick
Asking for guidance which library is required to get UHRR kicking in.

merci bien par avance, 73 de oe6rke

f4cib, 2020-11-29:
Merci, Thanks
As rkiendl i now struggle with rtlsdr required by Hamlib, I think it is related to PANadapter:
```
pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ ./UHRR
Traceback (most recent call last):
  File "./UHRR", line 19, in <module>
    from rtlsdr import RtlSdr
ModuleNotFoundError: No module named 'rtlsdr'

```
If i check the conf file nothing is related to rtlsdr, i've connected a perseus on my Raspberry, but up to now cannot find it with a lsusb on ttyUSBxx

```
pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ ls
LICENSE  README.md  UHRH.key  UHRR.conf  UHRR_users.db
opus     UHRH.crt   UHRR      UHRR.log   www
pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ cat UHRR.conf
[SERVER]
port = 8888
certfile = UHRH.crt
keyfile = UHRH.key
auth =
cookie_secret = L8LwECiNRxq2N0N2eGxx9MZlrpmuMEimlydNX/vt1LM=
db_users_file = UHRR_users.db
pam_account = pi
log_file = UHRR.log
debug = True

[CTRL]
interval_smeter_update = 0.5
debug = True

[AUDIO]
outputdevice = plughw:CARD=U0x41e0x30d3,DEV=0
inputdevice = plughw:CARD=U0x41e0x30d3,DEV=0

[HAMLIB]
rig_pathname = /dev/ttyUSB0
retry = 5
rig_model = FT817
trxautopower = True
rig_rate = 38400

[PANADAPTER]
sample_rate = 960000
center_freq = 68330000
freq_correction = 1
gain = 10
fft_window = hamming

pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $

``````
Is there any other file i have ton configure ? I checked that Perseus was supported on the riglist by Hamlib 
Merci 73 de Franck

F4HTB, 2020-11-29:
Hello, 

I have update the installation instruction for the new rtl sdr component.
Please run this:

sudo apt install rtl-sdr
sudo pip3 install pyalsaaudio pam pyrtlsdr

Sorry, i dont have any perseus for testing.
But maybe you can try somethings for debug.

`ls /dev/ | grep tty`
You have the list of tty. Then plug the perseus and run the command to.
Compare the two. 
One more should appear on the second command.
When you go ont he configuration page you will see it on the "rig_pathname " list.

For go on the config page you can access with this url:
https://raspberrypi.local:8888/CONFIG
Assuming you have not change the hostname of your pi.

In the conf file the PANADAPTER section is about rtl sdr dongle or other compatible with the rlt lib.

Let me know if it works.
73 F4HTB

f4cib, 2020-11-30:
Thanks/Merci Olivier,
Robert de OE6RKE, you should be able to make the server work if you follow these guidelines.
Server is now up here, i still need to deal with soundcard & TRX interfaces.
Keep you noticed.
Best regards,
73 de Franck, F4CIB

closed 2020-11-30 by F4HTB

f4cib, 2020-12-02:
Hi Olivier, 
Is there a way to add some serial configuration on the config page (ie. rts_state, serial_handshake, stop_bits) for RIG connection through Hamlib ?
Connection to rig using server is not working although i can read/write CAT to my FTDX5000 ?
`
pi@raspberrypi:~ $ rigctld -V
rigctl(d), Hamlib 4.1~git
`
I assume the connection is done by sending rigctld -m 1032 -r /dev/ttyUSB1 -s 9600 which is not enough in my case

The only way to access my FTDX is adding the following configuration options on serial port:
`rigctl -m 1032 -r /dev/ttyUSB1 -s 9600 --set-conf="serial_handshake=None" --set-conf="rts_state=OFF" --set-conf="stop_bits=2" -vvvvv`
Otherwise, i cannot initiate a communication with transceiver.

I add the same issue with [WebRig from OK1ZIA](http://tucnak.nagano.cz/wiki/WebRig) and i succeeded to access my TRX by adding the options in the php connection script.
May be i should open a new issue or request ?
73 de Franck, F4CIB

F4HTB, 2020-12-02:
Hello,

Can you add theyses lines  on line 343:

             self.rig.set_conf("dtr_state", "0")
             self.rig.set_conf("rts_state", "0")
             self.rig.set_conf("stop_bits", "2")
             self.rig.set_conf("serial_handshake", "None")

as fallow:

             self.rig = Hamlib.Rig(Hamlib.__dict__[self.rig_model])  # 
Look up the model's numerical index in Hamlib's symbol dictionary.
             self.rig.set_conf("rig_pathname", self.rig_pathname)
             self.rig.set_conf("retry", config['HAMLIB']['retry'])
             self.rig.set_conf("dtr_state", "0")
             self.rig.set_conf("rts_state", "0")
self.rig.set_conf("stop_bits", "2")
self.rig.set_conf("serial_handshake", "None")
             self.rig.open()

test and say me what append?

f4cib, 2020-12-02:
Bonjour Olivier,
Pour une raison inconnue ca ne marchait pas au début, puis après un reboot c'est OK.
A noter que j'ai mis le serial_handshake en premier, je ne sais pas si ca a le moindre intéret...
Ou puis-je trouver de la doc sur les fonctions self.rig.set_conf ?
Je présumes que c'est du Hamlib car je n'ai rien trouvé dans le fichier des scripts JAVA ni dans le reste du fichier qui decrivait la fonction... j'en conclue que c'est donc un appel à l'api Hamlib, mais je ne trouve rien dans les docs à ce sujet (j'ai peut-etre aussi mal cherché...)

Donc pour conclure plutôt que bricoler le fichier proposer d'agir sur ces paramètres dans le fichier de conf me semblerait une bonne chose. ce qui m'est m'étonne c'est que je suis le premier à remonter ce souci !  
Je poursuis mes tests, merci !

F4HTB, 2020-12-02:
Bonjour,

Je viens d'uploader une nouvelle version avec les options.
Testé chez moi mais reste à voir.

J'ai eu beaucoup de mal à trouver les infos.
Sous python j'ai navigué avec la fonction dir() et j'ai regardé du côté 
des api C de la page de référence.

Il y a eu beaucoup de dev et peu de man sur hamlib.

Effectivement, la hamlib possède des valeur par défaut pour chaque 
postes apparement.
Aussi, si vous n'avez pas celles d'usine ou que la hamlib n'a jamais été 
correctement renseigné, il se peu qu'il y ai des problèmes.
L'autre point important est que, suivant le câble cat que vous avez et 
son driver, parfois rtl ou dtr donnent des niveau haut ou bas par défaut.

Pouvez vous télécharger la dernière version du git et tester?

f4cib, 2020-12-03:
Bonsoir,
Desolé je reponds tardivement essayé ce matin j'ai cloné ca ne marchait pas (frequence farfelue 528294172 !) je reboote, pas mieux, je rajoute 8 bits none none 2 RTS & DTR OFF, et toujours pas mieux...le flux audio est ok mais visiblement la com cat est brouillée et ce dans les deux sens...
Tu veux que je testes quelque chose ?

F4HTB, 2020-12-03:
Bonsoir,

J'ai peut-être perdu le fils mais peux-tu me rappeler quel est ton 
installation?
Car il me semble qu'on a discuté d'un perseus mais je voie que tu as mis 
FTDX5000 comme model de rig.
Dans la liste des modèles il y a PERSEUS.

f4cib, 2020-12-03:
Bonsoir
je n'ai pas réussi a faire marcher le Perseus (en fait l'interface ttyUSB n'apparait pas dans la liste des devices.
J'arrive pourtant a dialoguer en ligne de commande avec les paquets distribués par Nico de Microtelecom.
Passons, je m'y re-essaierais plus tard.

Mon objectif reste cependant une émission/réception en remote basse consommation (donc raspberry ou équivalent) ce que permet ta ta solution.
J'ai donc branché mon FTDX5000 via une interface usb/rs232 5V de FTDI et dans les réglages de mon FT5000, je n'active pas le handshake RTS dans le menu de configuration. (rétrocompatibilité avec d'autres logiciels)

F4HTB, 2020-12-03:
J'ai beau y réfléchir, je n'ai pas de boule de cristal....Malheureusement...
Alors, on va commencer par essayer avec rigctl.
Peux-tu essayer ceci et vérifier que cela change bien la fréquence sur 
28500000:
Il faudra adapter 38400 au baut rate configuré chez toi.
Dit moi ce qui fonctionne et pas.

Ici on va vérifier si les paramètres par défaut fonctionnent:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 F 28500000

ensuite ici on test si mettre le dtr et rts à off permet de mieux 
fonctionné vis a vis de ton problème d'activation du cat:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 --set-conf="dtr_state=OFF" 
--set-conf="rts_state=OFF" --set-conf="serial_handshake=None" F 28400000

ensuite ici on rajoute un stop bit et on désactive le handshake:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=1" 
--set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" 
--set-conf="serial_handshake=None" F 28300000

ensuite si ca marche pas on test s'il faut deux stop bit et on désactive 
le handshake:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=2" 
--set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" 
--set-conf="serial_handshake=None" F 28200000

ensuite et ici on test en désactivant la parité:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=2" 
--set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" 
--set-conf="serial_handshake=None" --set-conf="serial_parity=None" F 
28100000

ensuite la même mais avec un seul stop bit:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=1" 
--set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" 
--set-conf="serial_handshake=None" --set-conf="serial_parity=None" F 
28100000

Enfin un dernier au cas où, on va forcer le 8 bits:
rigctl -m 1041 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=1" 
--set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" 
--set-conf="serial_handshake=None" --set-conf="serial_parity=None" 
--set-conf="data_bits=8" F 28750000

f4cib, 2020-12-04:
Bonjour,
Seule la première ne fonctionne pas:
```
pi@raspberrypi:~ $ rigctl -m 1032 -r /dev/ttyUSB0 -s 38400 F 28500000
set_freq: error = Feature not implemented
pi@raspberrypi:~ $ rigctl -m 1032 -r /dev/ttyUSB0 -s 38400 --set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" --set-conf="serial_handshake=None" F 28400000
pi@raspberrypi:~ $ rigctl -m 1032 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=1" --set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" --set-conf="serial_handshake=None" F 28300000
pi@raspberrypi:~ $ rigctl -m 1032 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=2" --set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" --set-conf="serial_handshake=None" F 28200000
pi@raspberrypi:~ $ rigctl -m 1032 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=2" --set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" --set-conf="serial_handshake=None" --set-conf="serial_parity=None" F 28100000
pi@raspberrypi:~ $ rigctl -m 1032 -r /dev/ttyUSB0 -s 38400 --set-conf="stop_bits=1" --set-conf="dtr_state=OFF" --set-conf="rts_state=OFF" --set-conf="serial_handshake=None" --set-conf="serial_parity=None" F 28000000
pi@raspberrypi:~ $
```
73 de Franck

f4cib, 2020-12-04:
Bon j'ai trouvé le problème...
La configuration est case sensitive.
Les informations serial_parity et serial_handshake doivent passer None et non NONE.
En modifiant manuellement le fichier UHRR.conf de la sorte, ca marche !

J'ai donc cherché sur le script Python où se faisait la configuration.
Et, en modifiant les les lignes 624 à 644 de la sorte ca devrait marchait (il restait d'ailleurs un copier/coller ODD)
```
		self.write("""HAMLIB serial parity:<select name="HAMLIB.serial_parity">""")
		if(config['HAMLIB']['serial_parity']!=""):
			self.write("""<option value="""+config['HAMLIB']['serial_parity']+""" selected>"""+config['HAMLIB']['serial_parity']+"""</option>""")
		self.write("""<option value=\"\"></option>""")
		self.write("""<option value=\"None\">None</option>""")
		self.write("""<option value=\"Odd\">Odd</option>""")
		self.write("""<option value=\"Even\">Even</option>""")
		self.write("""<option value=\"Mark\">Mark</option>""")
		self.write("""<option value=\"Space\">Space</option>""")
		self.write("""</select> Leave blank to use the HAMIB default value.<br/><br/>""")

		self.write("""HAMLIB serial handshake:<select name="HAMLIB.serial_handshake">""")
		if(config['HAMLIB']['serial_handshake']!=""):
			self.write("""<option value="""+config['HAMLIB']['serial_handshake']+""" selected>"""+config['HAMLIB']['serial_handshake']+"""</option>""")
		self.write("""<option value=\"\"></option>""")
		self.write("""<option value=\"None\">None</option>""")
		self.write("""<option value=\"XONXOFF\">XONXOFF</option>""")
		self.write("""<option value=\"Hardware\">Hardware</option>""")
		self.write("""</select> Leave blank to use the HAMIB default value.<br/><br/>""")
```
au redemarrage du serveur ca donne dans le Verbose:
```
HAMLIB rig_model FTDX5000
HAMLIB rig_pathname /dev/ttyUSB0
HAMLIB rig_rate 38400
HAMLIB trxautopower True
HAMLIB data_bits 8
HAMLIB stop_bits 2
HAMLIB serial_parity None
HAMLIB serial_handshake None
HAMLIB dtr_state OFF
HAMLIB rts_state OFF
```
Confirmé après un reboot, ca fonctionne.
Par contre je proposerais dès le démarrage du serveur de faire une requête de la fréquence car elle ne se met à jour qu'après avoir cliqué sur le vernier de fréquence de UHRR (Affichage 0 sur tous les digits). (d'ailleurs je ne sais pas si le mode se met à jour)

73 de Franck

f4cib, 2020-12-04:
J'ai soumis une pull-request ma 1ere ! j'espere que j'ai fait cela de la bonne manière !
73 de Franck

F4HTB, 2020-12-04:
Salut!

Super, merci d'avoir pris du temps pour tester et me confirmer cela.
Il est parfois difficile d'aider les gens sans avoir le matériel sous la 
mains...

Si ce n'est pas trop demandé, par hasard, tu n'aurais pas un amis qui à 
un IC-7300?
Il semble qu'il y ai un problème avec la carte son intégrée.
Mais il est difficile pour moi de trouver cela en temps de confinement :-)

En tout cas, merci encore une fois, ca fait plaisir d'avoir des 
personnes qui font des contributions pour faire évoluer ce petit projet.

PS: Je vais encore rajouter dans la page du conf quelque chose pour 
jouer sur les buffers audio.
Il semble en effet que lorsque les conditions ne sont pas parfaites, il 
y a encore des problèmes à ce niveau là.

73!

f4cib, 2020-12-05:
> Salut! Super, merci d'avoir pris du temps pour tester et me confirmer cela. Il est parfois difficile d'aider les gens sans avoir le matériel sous la mains... Si ce n'est pas trop demandé, par [...]

Bonjour, Avec plaisir, 
Malheureusement pour le 7300 personne à proximité, désolé !
je peux tester le FT847 et TS850 ainsi que quelques SDR si j'arrive à les faire détecter sous linux !
73 et Merci à toi,
Franck, F4CIB

Xpertti, 2020-12-05:
I got this ModuleNotFoundError: No module named 'Hamlib' ??
How to get rid of this error?
"Comment se débarrasser de cette erreur?"

pi@raspberrypi:~ $ cd Universal_HamRadio_Remote_HTML5
pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ ./UHRR
Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
ModuleNotFoundError: No module named 'Hamlib'
pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR
recording...
HTTP server started.
Tornado current user:None
SERVER port 8888
SERVER auth
SERVER db_users_file UHRR_users.db
AUDIO outputdevice plughw:CARD=Device,DEV=0
AUDIO inputdevice plughw:CARD=Device,DEV=0
HAMLIB rig_model FT891
HAMLIB rig_pathname /dev/ttyUSB0
HAMLIB rig_rate 38400
HAMLIB trxautopower True
HAMLIB data_bits
HAMLIB stop_bits
HAMLIB serial_parity
HAMLIB serial_handshake
HAMLIB dtr_state
HAMLIB rts_state
PANADAPTER center_freq 68330000
PANADAPTER sample_rate 960000
PANADAPTER freq_correction 1
PANADAPTER gain 10
PANADAPTER fft_window hamming
pi@raspberrypi:~/Universal_HamRadio_Remote_HTML5 $ recording...
HTTP server started.
Tornado current user:None
new connection on AudioRXHandler socket.
new connection on AudioTXHandler socket.
new connection on ControlTRX socket.
getFreq:
Send to all: getFreq:32
getMode:
Send to all: getMode:
PING
PING
PING
setFreq:22
Send to all: getFreq:14200000
overrun
PING
73 de OH4RL/Pertti