F4HTB/Universal_HamRadio_Remote_HTML5 #28: FILE auth is broken

dblacknc, 2021-01-28:
I couldn't get FILE authentication working and started looking.

The split of user and password here as read from the user DB file, around line 748 of UHRR:

...
                                        if x[0]!="#":
                                                db=x.split(" ")
                                                if db[0] == user and db[1] == password:
...

...results in a CR (\n) at the end of password, so the comparison always fails.
As a workaround/possible fix, I added, just before the split of x into db:

x=x.strip('\n')

Now it works.
I edited the user DB file using vi, on Linux.  So each line has a ctrl-J at the end.

F4HTB, 2021-01-28:
Thanks, I will test this asap. :)

F4HTB, 2021-07-23:
issue is corrected

closed 2021-07-23 by F4HTB