F4HTB/Universal_HamRadio_Remote_HTML5 #29: Extremely low TX audio

dblacknc, 2021-01-28:
The UHRR server side is relaying audio to the rig at an extremely low volume.  It's so low I can't even see the power level on the transceiver (in USB mode).   The transceiver is keyed as expected, no modulation.  If I yell into the (browser client) mic, I can see just a blip on the PTT light for the SignaLink USB.   The devices chosen are plughw:CARD=CODEC,DEV=0 for both input and output.

I have JTDX (FT8 app) able to run on the same computer, same sound card (SignaLink USB), and all works fine. It's able to send transmit (mic) audio to my transceiver, at full power - no problem.  I can even switch back and forth between this and UHRR, running one at a time, and JTDX works normally each time.  The ALSA mixer level does not change between the working JTDX and not working (for TX audio) UHRR.

When I transmit through the browser using UHRR, the audio level in the web UI looks good.  Obviously the client browser and mic are working well.   Receive audio is also audible and clear.

**Expected behavior**
The TX audio level should be much higher, and I see no way to correct or adjust this.  

** Equipment **
Server OS: Kubuntu Linux 20.10
Audio hardware: SignaLink USB
Rig: Yaesu FT-897D
CAT is just a serial dongle and it's working fine.

**Libs versions**
Hamlib 3.3
Python 3.8.6
tornado==6.0.4
pyalsa==1.1.6
pyalsaaudio==0.9.0

Client OS - two tested, same behavior.  This doesn't appear to be a client problem anyway.
Kubuntu Linux 20.10 with Microsoft Edge Dev 89.0
Fedora Linux 39 with Opera 73.0

F4HTB, 2021-01-28:
Hello, 
Normally it is not possible to change the output volume of uhrr software in another way than alsamixer.
Since, on the server side, it comes out on the same scale as the input of your microphone.
Maybe something to change pulse audio?
I have only try UHRR on raspberry pi, so ....

F4HTB, 2021-01-28:
If you find your problem, thank to make a feedback.

dblacknc, 2021-01-28:
I also tried the pulse devices and there was several seconds of buffering on the RX audio.  I don't think that solved the low volume problem, either.

The mixer for the (USB) CODEC device I use works for JTDX, but not UHRR:

$ amixer -c 1 get PCM
Simple mixer control 'PCM',0
  Capabilities: pvolume pswitch pswitch-joined
  Playback channels: Front Left - Front Right
  Limits: Playback 0 - 128
  Mono:
  Front Left: Playback 110 [86%] [-18.00dB] [on]
  Front Right: Playback 110 [86%] [-18.00dB] [on]

I will try increasing it to 100% but even 86% should be high enough.
Seems like this may have to do with the USB audio hardware I use.  I will keep looking for a solution and share it here if found.

dblacknc, 2021-01-29:
I figured this out - not a UHRR problem, as I kind of expected.  My rig's audio input via the data connector is only active when in the "DIG" mode, called "PKTUSB" in Hamlib 3.3.   Hamlib support for the FT-897D is only partial, anyway

I'm going to look at wiring the PC sound card to the mic port.

closed 2021-01-29 by dblacknc