F4HTB/Universal_HamRadio_Remote_HTML5 #30: Error whilst installing

TheOtherStu, 2021-05-18:
Hi,
I notice this hasn't been updated in a while and I wonder if a library has been updated.

I get all the way to the end without any errors then get this. I've tried as a regular user and also using su.

_Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 670, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 583, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1043, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory_

F4HTB, 2021-07-23:
Two way for install hamlib:
-by apt-get
-by compilation from source
Do not use both at the same time

closed 2021-07-23 by F4HTB

hb9pae, 2021-07-29:
Hello 

Faced the same Error as User TheOtherStu 
Hamlib was compiled from Source (as in the Wiki) 

Traceback (most recent call last):
  File "./UHRR", line 18, in <module>
    import Hamlib
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 17, in <module>
    _Hamlib = swig_import_helper()
  File "/usr/local/lib/python3.7/site-packages/Hamlib.py", line 16, in swig_import_helper
    return importlib.import_module('_Hamlib')
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libhamlib.so.4: cannot open shared object file: No such file or directory
 
 I called : PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR

hb9pae, 2021-07-29:
Problem solved (sudo ldconfig)! 

ldd /usr/local/bin/rigctl   shows libhamlib.so.4 => not found

after sudo ldconfig   libhamlib.so.4 is available!

73 de Peter HB9PAE