F4HTB/Universal_HamRadio_Remote_HTML5 #31: ICOM CI-V Address Configuration Item

hb9pae, 2021-06-05:
Thank you for this nice project.

I'm trying to get ICOM ID-5100 working. 
Compiling and Installation on RPI3 worked so far.

After start of UHRR-Server i can select TRX ID-5100 and the related Interface,
but no ideaa where to put the ICOM CI-V Address (Parameter -c) into the config fields.

Using Hamlib on terminal is OK:
 rigctl -m3071 -r /dev/ttyUSB0 -c 0x8c 
works so far.
 
Thank you for any help.

73 de Peter, hb9pae

arthur326, 2021-07-04:
Hello, I'm not one of the contributors and have not used the project myself yet; however, what about using rigctld as a workaround? First, connect rigctld with all the required options to the radio. Then, tell UHRR to use model number 2 and specify address:port as the resource.

If you see this and have a moment to spare, I'm interested about the purpose of specifying the CI-V address. Does it enable additional hamlib commands to be used?

hb9pae, 2021-07-29:
Thank you, I will go with this solution

hb9pae, 2021-09-10:
Thank you arthur326

This solution, using rigctrld is easy and worked fine!
73 de Peter HB9PAE

closed 2021-09-10 by hb9pae