F4HTB/Universal_HamRadio_Remote_HTML5 #40: Frequency

N7JYS, 2022-03-12:
As of 3/12/2022a downloaded version of Universal_Hamradio_Remote  will no longer control or display the frequencies of my FT-891.

Older version archived on my computer work fine with some latency issues.  Please Fix.

Eric 
N7JYS

N7JYS, 2022-03-14:
> It's sound like a hamlib problem.
> Can you say me if any message is written in the console where you lunch uhrr?

uhrr is executed for the Universal_Hamradio_Remote_HTML5 directory. So its using hamlib library from python3-libham-lib2.
After starting the server I bring up a google chrome browser in windows 10 and connect to the server
via https:// <ipaddress> and hit the on button it says problem then start the program. I have no controll of frequency or side bands, all that works is audio. Frequency display all remain zero's.
If I start the server using  PYTHONPATH=/usr/local/lib/python3.7/site-packages:$PYTHONPATH ./UHRR then all the zero's display random numbers.
Hamlib rigctl works fine when trying to control my FT-891 with its commands. This make me belive hamlib is working ok. So I'm missing somthing. I will start over with fresh install and see if that fixes the issue. Thanks. 

Eric (N7JYS) Marshall's Communications

N7JYS, 2022-03-16:
I believe I found the issue with the frequency not changing. My installation was ok. What I'm findingout is in the UHRR.conf file if I fill in the information in the DTR RTS drop down menus with either yes or no this causes the UHRR program to not communicate with Hamlib. Not sure what's going with the python language in your script but It dosen't seem to like it if these boxes are fill in. Since this is running on a raspberry pi It would be nicer if I could use a GPIO pin for PTT as I run my audio into the back side of my FT-891. This will eliminate the use of a serial cable for DTR signal to trigger the PTT  on the back side of the radio. Thanks!
 Eric (N7JYS) Marshall's Communications

scannerbrightly, 2022-12-10:
> I believe I found the issue with the frequency not changing. My installation was ok. What I'm findingout is in the UHRR.conf file if I fill in the information in the DTR RTS drop down menus with [...]

Wow, I had the **opposite resolution** to this problem. When I connect my Kenwood TS-570D directly via a USB to serial connector, I was getting junk displayed in the frequency area and pushing a button like "USB" would just blink green, even when the command window output seemed to be correct. 

I had to put something in both the RTS drop down for it to work correctly, which it now dows.