F4HTB/Universal_HamRadio_Remote_HTML5 #41: rigctld / NETRIGCTL

Links2004, 2022-04-17:
Hi, 
I am currently trying to get my X6100 running with UHRR.
but It looks like that UHRR can not handle the HAMLIB radio model `NETRIGCTL` correctly.
the dropdown still shows the tty/serial interfaces but for NETRIGCTL entering a IP is needed (in my case 127.0.0.1).
manually editing the config results in a python error.

```
[HAMLIB]
rig_pathname = 127.0.0.1
retry = 5
rig_model = NETRIGCTL
trxautopower = True
rig_rate = 38400
data_bits = 
stop_bits = 
serial_parity = 
serial_handshake = 
dtr_state = 
rts_state = 
```

```
$ ./UHRR
(<class 'TypeError'>, TypeError('function takes at most 5 arguments (7 given)'), <traceback object at 0xb41496c8>)
HTTP server started.
```

any hints on how to set up UHRR to work with a running rigctld?
tested with latest git (31fade18e6352fe6af91b1bbb6f43867a4498591)

vk4tux, 2022-11-05:
I have the same issue. I have rigctrld running on the windows FTDX101MP host, and can use node red etc to access, but not this ?

[SERVER]
port = 8881
certfile = UHRH.crt
keyfile = UHRH.key
auth = 
cookie_secret = L8LwECiNRxq2N0N2eGxx9MZlrpmuMEimlydNX/vt1LM=
db_users_file = UHRR_users.db
pam_account = vk4tux
log_file = UHRR.log
debug = True

[CTRL]
interval_smeter_update = 0.5
debug = True

[AUDIO]
outputdevice = plughw:CARD=U0x41e0x30d3,DEV=0
inputdevice = plughw:CARD=U0x41e0x30d3,DEV=0

[HAMLIB]
rig_pathname = 192.168.0.19
retry = 5
rig_model = NETRIGCTL
trxautopower = True
rig_rate = 38400
data_bits = 
stop_bits = 
serial_parity = 
serial_handshake = 
dtr_state = 
rts_state = 

[PANADAPTER]
sample_rate = 960000
center_freq = 68330000
freq_correction = 1
gain = 10
fft_window = hamming

scannerbrightly, 2022-12-04:
I also would like this issue resolved. 

I noticed that if you enclose the network address in single quotes, like '127.0.0.1', the program will not crash, but it still doesn't work to control my radio. Does that work for either of you?

F4HTB, 2022-12-04:
Hello Dear Om,

This is not a problem.
Indeed, UHRR is based on the use in direct cat mode of the stations by using the hamlib on serial.
Your request is therefore an addition of functionality.
However, if one of you wishes to become a contributor, it is with pleasure.
It will be enough to push your modifications and I would gladly accept them after analysis.

Sincerely, 73 F4HTB Olivier

closed 2022-12-04 by F4HTB