F4HTB/Universal_HamRadio_Remote_HTML5 #43: 7 arguments vs 5

vo1pwf, 2024-03-24:
got the latest software and ran it...as soon as I ran it the console showed 
:~/Downloads/hamradio_remote/Universal_HamRadio_Remote_HTML5$ ./UHRR 
(<class 'TypeError'>, TypeError('function takes at most 5 arguments (7 given)'), <traceback object at 0x7fa69ebeca00>)

trying to configure / save and run the software seems to get stuck on this 

tested on 3 pi's and a linux mint box

vo1pwf, 2024-03-24:
also on the bottom of the web page for config it shows same

Possible problem:(, TypeError('function takes at most 5 arguments (7 given)'), )

UNISTRAPHIOlivier, 2024-03-24:
Hello Dear, 

Generally this error comes from the fact that you are using the python3-pyalsaaudio library installed with apt-get.
You must use the updated version by pip
Please remove python3-pyalsaaudio and install like:
```
apt install python3-tornado python3-serial python3-pip python3-dev libasound2-dev
pip3 install pyalsaaudio
```

F4HTB, 2024-03-24:
Hello Dear,

Generally this error comes from the fact that you are using the python3-pyalsaaudio library installed with apt-get.
You must use the updated version by pip
Please remove python3-pyalsaaudio and install like:
```
apt install python3-tornado python3-serial python3-pip python3-dev libasound2-dev
pip3 install pyalsaaudio
```

vo1pwf, 2024-03-26:
BINGO!
sorryfor delay in replying
thanks for being soo quick on the help

looking forward to playing with the software over the long weekend

closed 2024-05-31 by F4HTB